cncgoko/Goko #12: Motor settings in TinyG configuration do not persist.

fzbxvudu, 2015-08-12:
When trying to modify the motor settings (for instance, trying to configure a dual Y motor setup) in the TinyG configuration panel, changes are not made nor do edits persist when the panel is reopened.  At this point, motor outputs cannot be assigned to correct axes nor can their characteristics be modified to calibrate the machine.

cncgoko, 2015-08-12:
Hello, 
Thank you for your feedback.
Goko doesn't persist TinyG configuration. It uses the connected TinyG controller and request for its configuration. 
Few things to check : 
- Is your TinyG board powered and connected ?
- Are you able to talk to your TinyG ? (you can enable the serial console view and see if TinyG talks back)
- When applying your settings, do you see any outgoing data in the Serial console view ?

You may have to enable the console view before doing anything to see the logged serial data.

fzbxvudu, 2015-08-13:
Here's what I can verify:
TinyG is powered and connected.  Requests to motors are processed and motors rotate.
TinyG communicates. 
After configuring motors to proper axes (i.e. Motor 1 - X, Motors 2 & 3 - Y, Motor 4 - Z) and pressing OK, there is no data appearing in the Serial console view.  I taking this to mean that no configuration information is being sent along to the TinyG.

Let me know if you need additional information.  Thanks for quick response to issue posting!

cncgoko, 2015-08-13:
You're right, I managed to reproduced the issue. This will be fixed in v0.0.2 which should be available soon.

For now, I recommend you use the Serial console to configure your dual motor as described in the TinyG wik ( [Motor mapping doc](https://github.com/synthetos/TinyG/wiki/TinyG-Configuration-for-Firmware-Version-0.97#1ma---map-motor-to-axis)) i. In your case :
```
$1ma=0
$2ma=1
$3ma=1
$4ma=2
```

closed 2015-09-21 by cncgoko