cncgoko/Goko #13: Flow control ($ex)

electric-2000, 2015-09-20:
Nice work!! Glad to see some work going on with GOKO, this will be the first offline software to work with the TinyG.

I've upgraded to your latest and I'm not able to send a file. I get a "No execution control enabled." error.

TinyG with the latest 440.2 firmware
Goko Alpha 0.0.2 

[image]

electric-2000, 2015-09-21:
O.K. I figured out the issue. In the new version you have "Serial Connection Settings" & the TinyG "Configuration" dropdown. I have synced the Device Manage settings on the com port with the settings in Goko.
We are now cutting carbon plates!
Just checked and the Motor settings are sticking now. One thing I did is hit the reset after sending the changes.

Thanks for a great program!!
-Mike

cncgoko, 2015-09-21:
Hi Mike,

Thanks for the feedback. Glad to see someone is using it :)
Goko makes sure there is at least one 'execution control' enabled. Either flow control (Xon/Xoff or Rts/Cts) or Queue report.
I added a check in TinyG to make sure that the TinyG $ex and Serial connection match.
I fixed a few issues this past weekend, but didn't push any update yet.
You should be able to do (and test :) ) an auto update when it's available.

I also opened issues on TinyG regarding arcs (G2 and G3) :
 * [ Arc command (G2 & G3) first status report is wrong #140 ](https://github.com/synthetos/TinyG/issues/140)
 * [ G3 command make arc + 1 full revolution and wrong direction #141 ](https://github.com/synthetos/TinyG/issues/141)

I guess I can close this issue ?
Feel free to join on [Gitter](https://gitter.im/cncgoko/Goko). I'm connected as much as possible.

electric-2000, 2015-09-21:
Glad to see you finishing up Goko, I was hoping that the program would not go stale! With most of the TinyG owners using Chilipeppr and wanting them to go offline, I can’t believe there are not more switching over to Goko. Now that you have something that is working reliably, I’ll be doing some posting with my new setup. We are a startup company manufacturing CNC routers, 3D Printers and desktop vacuum formers. I have been searching for months trying to find software for either a GRBL, Arduino or the TinyG to work with our machines. I ran across Goko on the forums a few months back and have been using it, but the DRO did not give feedback and that made it frustrating.

As for Arcs, the software that I draw with does not convert properly so I have not been using them. I use Vectric Cut2D with a custom Post Processor that I modified and have been successful with that. I keep my drawings simple, using Illustrator, Corel Draw or InkScape and export EPS files.

In the attachment you can see how I have mounted the TinyG on the X gantry, this helps keep the wiring to a minimum.

-Mike

cncgoko, 2015-09-21:
Nice machine.
Goko is not finished. I'm not sure it will be one day :)
I still have some new functionalities planned.

The main reason why few people switch to Goko is because I don't "advertise" much for now. I'd like to stabilize the TinyG and Grbl features. It still requires testing/tuning. For instance, I didn't post the v0.0.2 in any forum because I'd like to test Grbl first (which I can't because I don't use it anymore :) ).

electric-2000, 2015-09-21:
I have a couple Grbl boards that I have been testing, an Arduino UNO “CNC Shield” and a Smoothieboard. Of the two, I’d prefer to go with the Smoothie which has feature sets for both CNC and 3D printing.

There are issues with Goko switching between the TinyG and Grbl-
--The communication settings are not sticky
--When you change the Target Board in the Preferences, you need to restart Goko to proceed with the new selection.
--The Smoothiboard is not recognized, Grbl is looking for “Hello Grbl” and Smoothie is displaying “Hello Smoothie”

Feature Wishlist-
--Support for ShuttlExpress jog controller
--Custom Macros

cncgoko, 2015-09-21:
The fact is that even if Smoothie is 100% compatible with Grbl, it certainly has it's own specificity. Support for Smoothie is not planned in immediate future. Plus, I do not own a Smoothie so developping for it will be difficult.

Macro is something I've been thinking to as well. Can you please describe a little more your needs, and what you have in mind ?

electric-2000, 2015-09-21:
I use a program for 3D printing (MatterControl) that has Macro capability, this is a screenshot of the Editor-

A couple examples of macros that I could use is Z-Probing  and controlling an external fan.

[cid:image001.png@01D0F460.F6239830]

cncgoko, 2015-09-21:
I see.

BTW the restart when switching target board (Grbl/TinyG) is normal and required. But maybe adding a message or even an automatic restart would help.

closed 2017-05-28 by cncgoko