cncgoko/Goko #17: 64bit install

electric-2000, 2015-09-23:
I have a  newly configured 64 bit machine and I cant install the latest build, I get a JAVA error.
-Windows Home Premium 64 bit
-8 gigs ram
-Java 8 installed
The 32 bit installs and runs just fine...
[image]

Another couple issues-
-Keyboard Jog enabled does not appear to work *if it did I could possible hook up a ShuttlExpress)
-In the TinyG config we need a checkbox to enable/disable the "Soft Limits". I discovered with it enabled it has caused some issues with the DRO not working and no limit errors showing up.

cncgoko, 2015-09-23:
**Launch error**
Make sure you have a 64bits Java installed. To check this:
- run a ```cmd``` in windows 
- enter the command ```java -d64 -version```. 

If the result is something like ```This Java instance does not support a 64-bit JVM. Please install the desired version.``` then you do have a 32bits JVM installed. In this case, the command ```java -d32 -version``` should display the correct 32bits version number

To run the 64bits version of Goko, you have to install the 64 bits JVM (which is recommended under 64bits OS)

**Keyboard jog**
The keyboard jog works only if the 3D viewer has focus and when the red warning is displayed. This was mainly to avoid unexpected jog when using keyboard arrow keys in textfields. I have to find a better way to handle this.

**Soft limits**
Are you talking about the ```$sl``` parameters ? I mainly developped using fb 335.** of TinyG. I don't remember it being available. I'll include it as well as the alarm state. I'll also have to check for the change between the firmware build

electric-2000, 2015-09-23:
You are correct, I assumed that Java would have detected the 64bit OS. Installed and working.

I see your issue with the keyboard Jog, instead of the red warning you could make a modal window with a DRO and on/off toggle. When active the jog will always be live.

Including an alarm state would have helped me in the past. Adding the toggle in the config would also be great. The Soft Limits code reads-  “  [sl]  soft limit enable           0”

cncgoko, 2015-09-24:
That's an idea for the jog feature. I'll have to do experiment on how to make more easy and natural to use.

Actually the alarm state already exist (since the beginning). I guess it was never sent by TinyG. I never really tried soft limits. That's actually an interesting functionality.

electric-2000, 2015-09-24:
I forgot to mention, you are missing a spinner for the Step Size, it could go under the Speed spinner in the command panel.

cncgoko, 2015-09-24:
The step size is only active when running Grbl. TinyG works with continuous jog. I guess it could support step jog as well.

electric-2000, 2015-09-25:
Yes, the automatic update worked. 
Don't get rid of the continuous jog, that is a great feature. I like setting a step size so that I can zero out the machine then start at 10mm in from the XY, the step jog just makes it a few less steps. This would be another great macro, once it gets implemented.

cncgoko, 2015-09-25:
Getting rid of the continuous/rapid jog is not planned. But I do have to find another way to achieve it now that soft limits can be enabled. Using the current way **(jog is actually a long move like G1X1000 and is stopped with ! when the jog button is released)**  always trigger the alarm...

commit c72d056a (2015-09-27): Fix #17
commit 46e43fc0 (2015-09-27): Fix #17
closed 2016-05-01 by cncgoko