cncgoko/Goko #18: TinyG g2 on Arduino Due

mpatoulachik, 2015-09-30:
Hi,
A (probably stupid) question:
I am testing Goko 0.0.2 with TinyG g2 on an Arduino Due and a gShield.
The firmware is v0.98 build 87.01 (grabbed and flashed from Chilipeppr).
So far I can connect to the board (parameters are read in the serial console when the connection is established) and jog all the axes, but nothing more. The TinyG configuration window doesn't see the board, it shows no version or ID and all parameters are at 0.
As far as I understood, the "original" TinyG and g2 are pretty much identical, except that g2 does not support parameters ec,ee,ex and baud.
Is it why I can't get it to work, or are there some more incompatibilities? 
Or am I doing something else wrong?

cncgoko, 2015-09-30:
Hi,
Can you please copy/paste the serial console content ?

mpatoulachik, 2015-09-30:
This is what I get in the serial console
- From connecting on COM13 (Control channel):
```
{"r":{"fv":0.98,"fb":87.01,"hp":3,"hv":0,"id":"0213-2335-5383","msg":"SYSTEM READY"},"f":[1,0,1]}
{"sr":""}
{"qr":""}
{"1":""}
{"2":""}
{"3":""}
{"4":""}
{"x":""}
{"y":""}
{"z":""}
{"a":""}
{"sys":""}
{"G55":""}
{"G56":""}
{"G57":""}
{"G58":""}
{"G59":""}
```
 - and from connecting on COM14 (data channel):
```
{"r":{"fv":0.98,"fb":87.01,"hp":3,"hv":0,"id":"0213-2335-5383","msg":"SYSTEM READY"},"f":[1,0,1]}
{"sr":""}
{"qr":""}
{"r":{"sr":{"line":0,"posx":0.000,"posy":0.000,"posz":0.000,"posa":0.000,"feed":0.00,"vel":0.00,"unit":1,"coor":1,"dist":0,"admo":1,"frmo":1,"momo":4,"stat":1}},"f":[1,0,10]}
{"1":""}
{"2":""}
{"r":{"qr":28},"f":[1,0,10]}
{"r":{"1":{"ma":0,"sa":1.800,"tr":40.0000,"mi":8,"po":0,"pm":2,"pl":0.375}},"f":[1,0,9]}
{"3":""}
{"r":{"2":{"ma":1,"sa":1.800,"tr":40.0000,"mi":8,"po":0,"pm":2,"pl":0.375}},"f":[1,0,9]}
{"4":""}
{"r":{"3":{"ma":1,"sa":1.800,"tr":40.0000,"mi":8,"po":1,"pm":2,"pl":0.375}},"f":[1,0,9]}
{"x":""}
{"r":{"4":{"ma":2,"sa":1.800,"tr":1.2500,"mi":8,"po":0,"pm":2,"pl":0.750}},"f":[1,0,9]}
{"y":""}
{"r":{"x":{"am":1,"vm":50000,"fr":50000,"tn":0.000,"tm":420.000,"jm":10000,"jh":20000,"jd":null,"hi":1,"hd":0,"sv":3000,"lv":100.00,"lb":20.000,"zb":3.000}},"f":[1,0,9]}
{"z":""}
{"a":""}
{"sys":""}
{"G55":""}
{"G56":""}
{"G57":""}
{"G58":""}
{"G59":""}
```

cncgoko, 2015-09-30:
If I understand correctly g2, when opening a single port, it becomes control and data port. I think that's waht you'll have to do since Goko doesn't handle 2 serial ports (TinyG only use a one port). Maybe you'll have to restart your board, or only have one port connected at a time (in all apps, not only Goko).

From what I see from your console, the dialog looks ok, except that it doesn't answer after ```{"x":""}``` request. Maybe a buffer overflow due to g2 not handling flow control (```$ex param```)

It could be an issue with the unhandled parameters actually. The configuration panel does require them I think. Is there anything (errors...) in the log files ?

mpatoulachik, 2015-09-30:
Ok, I'll try to get it to open only one port and see where it leads.
Github doesn't let me attach the logfile for some reason so the log is here:
http://pastebin.com/5rpQfP1Z

cncgoko, 2015-09-30:
Ok. You can also use Gist on Github to post your file (https://gist.github.com/)
Can you try an resend all the group request params one at a time in the serial console ?
Typically :
```
{"1":""}
{"2":""}
{"3":""}
{"4":""}
{"x":""}
{"y":""}
{"z":""}
```

Also make sure you have the following parameters like below : (you can set them using Serial console):
```
{"js":1}
{"jv":5}
```

It seems that g2 does not require flow control. I'll have to read more about it.

mpatoulachik, 2015-10-01:
I get no response at all in the serial console, whatever I type and independently from which port I connect to.
I connected the board to Chilipeppr to make sure that JS=1 and JV=5 an then retried in Goko, but no luck.
I also deactivated the data port (in the device manager) to force g2 to connect both data and control over one port, but no luck there either...
Seems that TinyG and G2 are not so similar after all.

cncgoko, 2015-10-01:
I found the issue and it's definitely in Goko. I'll have to tweak the way parameters are received a little bit, not only to support g2 but also to support missing or changes in the TinyG parameters in next releases.

cncgoko, 2015-10-02:
Can you please make sure that a null value is not a normal behavior ?
Can you retry after making sure there is no null value in the parameter ? (you'll have to use another serial software)

mpatoulachik, 2015-10-02:
Apparently the `jd` parameter is deprecated, but somehow still present (they use another cornering algorithm). As I understood it will always be null, whatever I try to write, G2 will silently discard the change.

This seems to be the case since ver. 085.06
https://github.com/synthetos/g2/commit/541bce483caaf878b56e67d91d24e6418fd4925a

As I understand there is a new cornering algorithm using only `ja` :
https://github.com/synthetos/g2/commit/229f71fba23701f3c5d7c3e45a5bd2cb5c8087ac

cncgoko, 2015-10-02:
I see. I pushed a quick temporary fix that you should be able to get by clicking the ```Help > Update``` button.
Basically it allows to read ```null``` parameters from TinyG and display them in configuration panel. The configuration panel will still requires a valid value in the ```$jd``` field. Simply enter any valid number and you will be allowed to validate. The value will be sent and TinyG will ignore it.

closed 2015-10-06 by cncgoko

cncgoko, 2015-10-07:
Any improvements with the latest update ?

mpatoulachik, 2015-10-09:
Yes it did help, I can now change the parameters in G2, so your update worked well.
Goko is still unable to send a Gcode file to G2, I guess because of the different flow control compared to TinyG: 
`No execution control enabled. Please use flow control ($ex) or queue report ($qr)`

Anyhow it's up to you if you want to take the fight with g2, I can stick to grbl to use Goko.

cncgoko, 2015-10-10:
Thanks for the update.
Running TinyG v0.97, it's almost mandatory to use the flow vontrol. That's why Goko checks and won't run the file if flow control is off. 
I'll have to think about adding support for g2, but not having a g2 board myself will be a difficulty. Glad you can use Grbl + Goko though.

cncgoko, 2015-10-13:
@mpatoulachik are you using Goko with Grbl on a "regular" basis ? Have you experienced any issues ? Do you think about any possible improvement ?
Thanks

aldenhart, 2015-10-13:
FYI: You can run g2 on an Arduino Due. a gShield get you 3 motor drivers. All 6 axes are available on the Due pinouts if you want more than 3.

cncgoko, 2015-10-13:
@aldenhart good point. Besides, since I'm only working on controlling the board on don't even need to have it tied to an actual motor driver. I only need the communication setup. I'll have to give it a try.

mpatoulachik, 2015-10-13:
I don't use GRBL on a regular basis yet, I am still building the mechanical part but have the electronics rigged (so i know I can at least move the steppers). I'll let you know if I find something as I test and use goko.

commit 08596168 (2015-10-06): Quick fix #18 [skip ci]

More advanced management of different TinyG versions is required.