cncgoko/Goko #19: G0 command

electric-2000, 2015-10-06:
Not sure if this is a TinyG issue or something in your code.
When running a file, I issue G0Z12.000 to bring the tool to the start point of the cut and when finished I use G0Z12.000 to clear the part to return the tool to home. On sample below the ending G0Z12.000 does not get executed, but the following line G0X0.000Y0.000 does.

G90
G17
G21
G0Z12.000
G0X0.000Y0.000
M03 S11000
G0X118.619Y33.587Z12.000
G1Z-3.000F600.0
G1X119.133Y31.929
G1X119.665Y30.301
...
...
...
...
...
G1X118.619Y33.587
G0Z12.000
G0X0.000Y0.000
M05

cncgoko, 2015-10-06:
HI,

I tried and streamed the given example to TinyG (only removed the ```...``` lines) and it works like a charm (using Goko 0.0.2 and TinyG 0.97 firmware build 440.18).
Anything in the log file ? 
You'll find 2 logs file :
- a .log file in the log/ folder
- or the most recent log file in workspace/.metadata/ folder

electric-2000, 2015-10-06:
Here is the log file and the entire gcode.

G1X120.214Y28.707
G1X120.782Y27.148
G1X121.369Y25.624
G1X121.975Y24.137
G1X122.600Y22.689
G1X123.245Y21.280
G1X123.910Y19.913
G1X124.595Y18.588
G1X125.302Y17.306
G1X126.030Y16.070
G1X126.779Y14.880
G1X127.551Y13.738
G1X128.345Y12.645
G1X129.162Y11.602
G1X130.002Y10.611
G1X130.866Y9.674
G1X131.755Y8.792
G1X132.668Y7.965
G1X133.604Y7.199
G1X134.085Y6.835
G1X134.570Y6.488
G1X135.061Y6.157
G1X135.559Y5.840
G1X136.063Y5.540
G1X136.574Y5.255
G1X137.092Y4.986
G1X137.616Y4.734
G1X138.146Y4.498
G1X138.684Y4.279
G1X139.227Y4.076
G1X139.778Y3.891
G1X140.335Y3.723
G1X140.898Y3.572
G1X141.468Y3.439
G1X142.045Y3.324
G1X142.628Y3.226
G1X143.218Y3.147
G1X143.815Y3.087
G1X144.418Y3.044
G1X145.027Y3.021
G1X145.521Y3.017
G1Z0.000
G1X145.641Y3.016
G1X146.545Y3.030
G1X147.439Y3.084
G1X148.320Y3.180
G1X149.186Y3.315
G1X149.497Y3.379
G1Z-3.000
G1X150.037Y3.489
G1X150.874Y3.703
G1X151.697Y3.955
G1X152.506Y4.244
G1X153.300Y4.570
G1X154.079Y4.933
G1X154.844Y5.331
G1X155.595Y5.764
G1X156.332Y6.232
G1X157.054Y6.733
G1X157.763Y7.268
G1X158.457Y7.834
G1X159.138Y8.433
G1X159.805Y9.063
G1X160.459Y9.723
G1X161.100Y10.413
G1X161.727Y11.133
G1X162.342Y11.881
G1X162.944Y12.658
G1X163.533Y13.462
G1X164.110Y14.293
G1X164.675Y15.151
G1X165.227Y16.035
G1X165.768Y16.944
G1X166.297Y17.878
G1X166.815Y18.836
G1X167.321Y19.818
G1X167.816Y20.823
G1X168.300Y21.851
G1X168.773Y22.901
G1X169.235Y23.973
G1X169.687Y25.066
G1X170.128Y26.180
G1X170.558Y27.313
G1X170.979Y28.466
G1X171.390Y29.639
G1X172.180Y32.036
G1X172.933Y34.504
G1X173.648Y37.037
G1X174.326Y39.631
G1X174.969Y42.281
G1X175.578Y44.983
G1X176.153Y47.734
G1X176.696Y50.527
G1X177.207Y53.360
G1X177.688Y56.228
G1X178.140Y59.126
G1X178.563Y62.050
G1X178.960Y64.997
G1X179.330Y67.960
G1X179.674Y70.937
G1X179.995Y73.923
G1X180.292Y76.913
G1X180.566Y79.903
G1X180.820Y82.889
G1X181.053Y85.866
G1X181.267Y88.831
G1X181.462Y91.778
G1X181.640Y94.703
G1X181.802Y97.603
G1X182.081Y103.305
G1X182.306Y108.852
G1X182.486Y114.207
G1X182.628Y119.336
G1X182.741Y124.203
G1X182.832Y128.775
G1X182.981Y136.888
G1X183.055Y140.359
G1X183.140Y143.408
G1X183.112Y144.635
G1Z0.000
G1X183.070Y146.463
G1X183.036Y148.635
G1Z-3.000
G1X183.016Y149.939
G1X182.923Y158.060
G1X182.789Y167.503
G1X182.686Y172.635
G1X182.547Y177.992
G1X182.363Y183.541
G1X182.127Y189.246
G1X181.829Y195.072
G1X181.654Y198.020
G1X181.461Y200.986
G1X181.248Y203.964
G1X181.014Y206.951
G1X180.758Y209.943
G1X180.479Y212.934
G1X180.176Y215.921
G1X179.848Y218.900
G1X179.494Y221.865
G1X179.113Y224.813
G1X178.703Y227.739
G1X178.264Y230.639
G1X177.794Y233.509
G1X177.293Y236.344
G1X176.758Y239.140
G1X176.190Y241.893
G1X175.587Y244.598
G1X174.948Y247.252
G1X174.272Y249.849
G1X173.557Y252.385
G1X172.803Y254.855
G1X172.411Y256.067
G1X172.008Y257.259
G1X171.595Y258.433
G1X171.172Y259.588
G1X170.737Y260.724
G1X170.292Y261.840
G1X169.835Y262.935
G1X169.368Y264.009
G1X168.889Y265.062
G1X168.398Y266.092
G1X167.896Y267.100
G1X167.382Y268.085
G1X166.856Y269.046
G1X166.318Y269.983
G1X165.767Y270.895
G1X165.204Y271.781
G1X164.628Y272.642
G1X164.040Y273.476
G1X163.438Y274.283
G1X162.823Y275.063
G1X162.194Y275.815
G1X161.552Y276.538
G1X160.896Y277.231
G1X160.226Y277.895
G1X159.542Y278.528
G1X158.844Y279.130
G1X158.131Y279.700
G1X157.404Y280.238
G1X156.662Y280.743
G1X155.904Y281.215
G1X155.132Y281.652
G1X154.345Y282.054
G1X153.543Y282.420
G1X152.726Y282.750
G1X151.893Y283.044
G1X151.045Y283.300
G1X150.182Y283.518
G1X149.304Y283.697
G1X148.410Y283.837
G1X147.502Y283.937
G1X146.578Y283.997
G1X145.639Y284.016
G1X145.260Y284.008
G1Z0.000
G1X144.701Y283.997
G1X143.777Y283.937
G1X142.868Y283.837
G1X141.975Y283.697
G1X141.293Y283.558
G1Z-3.000
G1X141.097Y283.518
G1X140.233Y283.300
G1X139.386Y283.044
G1X138.553Y282.750
G1X137.736Y282.420
G1X136.933Y282.054
G1X136.146Y281.652
G1X135.374Y281.215
G1X134.617Y280.743
G1X133.875Y280.238
G1X133.148Y279.700
G1X132.435Y279.130
G1X131.736Y278.528
G1X131.052Y277.895
G1X130.383Y277.231
G1X129.727Y276.538
G1X129.084Y275.815
G1X128.456Y275.063
G1X127.841Y274.283
G1X127.239Y273.476
G1X126.650Y272.642
G1X126.074Y271.781
G1X125.511Y270.895
G1X124.961Y269.983
G1X124.422Y269.046
G1X123.896Y268.085
G1X123.382Y267.100
G1X122.880Y266.092
G1X122.390Y265.062
G1X121.911Y264.009
G1X121.443Y262.935
G1X120.987Y261.840
G1X120.542Y260.724
G1X120.107Y259.588
G1X119.684Y258.433
G1X119.271Y257.259
G1X118.868Y256.067
G1X118.475Y254.855
G1X117.722Y252.385
G1X117.007Y249.849
G1X116.331Y247.252
G1X115.691Y244.598
G1X115.088Y241.893
G1X114.520Y239.140
G1X113.986Y236.344
G1X113.485Y233.509
G1X113.015Y230.639
G1X112.576Y227.739
G1X112.166Y224.813
G1X111.784Y221.865
G1X111.430Y218.900
G1X111.102Y215.921
G1X110.800Y212.934
G1X110.521Y209.943
G1X110.265Y206.951
G1X110.031Y203.964
G1X109.818Y200.986
G1X109.625Y198.020
G1X109.450Y195.072
G1X109.293Y192.146
G1X109.152Y189.245
G1X108.915Y183.541
G1X108.732Y177.992
G1X108.593Y172.635
G1X108.490Y167.503
G1X108.414Y162.633
G1X108.356Y158.059
G1X108.266Y150.229
G1Z0.000
G1X108.262Y149.939
G1X108.209Y146.463
G1X108.204Y146.229
G1Z-3.000
G1X108.139Y143.409
G1X108.248Y139.280
G1X108.344Y134.421
G1X108.568Y122.792
G1X108.735Y116.184
G1X108.962Y109.157
G1X109.103Y105.511
G1X109.267Y101.790
G1X109.455Y98.005
G1X109.670Y94.166
G1X109.914Y90.282
G1X110.189Y86.365
G1X110.497Y82.423
G1X110.842Y78.467
G1X111.224Y74.508
G1X111.647Y70.555
G1X112.113Y66.618
G1X112.624Y62.708
G1X113.183Y58.834
G1X113.791Y55.008
G1X114.115Y53.114
G1X114.452Y51.237
G1X114.802Y49.376
G1X115.167Y47.533
G1X115.545Y45.710
G1X115.938Y43.907
G1X116.346Y42.125
G1X116.769Y40.367
G1X117.208Y38.632
G1X117.662Y36.924
G1X118.132Y35.242

cncgoko, 2015-10-06:
You can use Github Gist (https://gist.github.com/). I think you can add files in them.

electric-2000, 2015-10-06:
Here is the log file. I noticed that when the G0Z12.00 was executed there was beep coming from the board or stepper.
 
https://gist.github.com/06dded43ee739b7b6ebd.git

cncgoko, 2015-10-06:
Gist URL is not valid.

electric-2000, 2015-10-06:
That is the HTTPS clone URL that was supplied, go figure...try this
https://gist.github.com/electric-2000/06dded43ee739b7b6ebd

cncgoko, 2015-10-06:
The complete file run without issue on my side and there is nothing in the log file.
This could be a TinyG 440.20 as it's the edge. Current master is 440.18 and that's the version I'm using.
Any error in the most recent log file in  ```{GOKO_FOLDER}/workspace/.metadata/``` folder ?

electric-2000, 2015-10-06:
Here is link to my workspace .log file-
https://gist.github.com/electric-2000/ff3b70edfa35de761d45

Sorry to drag you into this, here is where I am-
FWIW- 440.20 is called a Master so I figured it was good.
Loaded Chilipeppr and had the same issue, so it's not GOKO
Downgraded to 440.18 and it's working, so it's a 440.20 has an issue, guess I should file a bug with them.

Thanks Again

aldenhart, 2015-10-06:
Please get me your machine configuration and I will check this out. Did it seem like the motor was trying to move on the G0 Z12 move?

electric-2000, 2015-10-06:
It's working great with 440.18, it's 440.20 that has the issue
Here is a link to the config-
https://gist.github.com/electric-2000/315abcfec95d7181c87b

electric-2000, 2015-10-06:
Here are a couple shots to the hardware configuration-
http://www.3dimensionaltools.com/wp-content/uploads/Komodo-Right-Wiring.jpg
http://www.3dimensionaltools.com/wp-content/uploads/Komodo-Right.jpg

aldenhart, 2015-10-06:
I've only had the chance to test in a cursory way but the Z axis seems to be working fine on both 440.18 and 440.20. I notice that your jerk settings are very high for a screw axis for Z. Screw axes are typically happiest when the jerk is set to between 50 (M) and 500 (M). You might try that.

Other things I noticed about the configs
- A motor timeout of 0.1 sec is very low for a belt driven machine. You usually want this at least 2 seconds so you don't get drift when motion stops
- The X and Y pulleys are very close but not the same. Usually these are the same. Were these adjusted for positioning?
- If the Z screw is a typically 1.25mm screw (stock Shapeoko) then the value of 1.24 is too low by a hair
- The feedrate maxes should be below the velocity maxes.

electric-2000, 2015-10-06:
I come from the Mach3 world and as you can tell my settings were all over the place! I lowered the Z jerk and brought the feedrates below the max velocity...that took care of the stall I was getting on the Z retract.
After testing in 440.18 I moved back up to 440.20 and the problem went away!
As for the travel settings, I did a calibration test early on and this is what I came up with, I'll need to revisit that again to make sure my moves are calibrated.

Thanks Alden for jumping in here and helping out.

closed 2015-10-07 by cncgoko