cncgoko/Goko #21: TinyG support

perfectphase, 2015-12-29:
Hi, I've just downloaded the 64bit version and it auto updated, but in the preferences the only target board is Grbl, how do I tell it to use the TinyG controller?

Cheers,

Stephen.

perfectphase, 2015-12-29:
Looking at the logs I found this
19:23:28.473 [org.goko.log.GokoLogConsoleReader@51a0aa70] INFO  OSGI Log -  [ Bundle = org.goko.controller.tinyg] InfoServiceEvent REGISTERED
19:23:28.477 [org.goko.log.GokoLogConsoleReader@51a0aa70] INFO  OSGI Log -  [ Bundle = org.goko.controller.tinyg] InfoServiceEvent UNREGISTERING
19:23:28.491 [org.goko.log.GokoLogConsoleReader@51a0aa70] INFO  OSGI Log -  [ Bundle = org.goko.controller.tinyg] InfoBundleEvent STOPPED
19:23:28.507 [org.goko.log.GokoLogConsoleReader@51a0aa70] INFO  OSGI Log -  [ Bundle = org.eclipse.osgi] ErrorAn error occurred while automatically activating bundle org.goko.controller.tinyg (109).
19:23:28.528 [org.goko.log.GokoLogConsoleReader@51a0aa70] INFO  OSGI Log -  [ Bundle = org.eclipse.equinox.ds] ErrorException occurred while creating new instance of component Component[
	name = org.goko.controller.tinyg.TinyGFeatureSet
	activate = activate
	deactivate = deactivate
	modified = 
	configuration-policy = optional
	factory = null
	autoenable = true
	immediate = false
	implementation = org.goko.controller.tinyg.TinyGFeatureSet
	state = Disposed
	properties = 
	serviceFactory = false
	serviceInterface = [org.goko.core.feature.IFeatureSet]
	references = {
		Reference[name = ISerialConnectionService, interface = org.goko.core.connection.serial.ISerialConnectionService, policy = static, cardinality = 1..1, target = null, bind = null, unbind = null]
		Reference[name = IGCodeService, interface = org.goko.core.gcode.service.IGCodeService, policy = static, cardinality = 1..1, target = null, bind = null, unbind = null]
		Reference[name = IGCodeExecutionMonitorService, interface = org.goko.core.gcode.service.IGCodeExecutionMonitorService, policy = static, cardinality = 1..1, target = null, bind = null, unbind = null]
		Reference[name = IApplicativeLogService, interface = org.goko.core.common.applicative.logging.IApplicativeLogService, policy = static, cardinality = 1..1, target = null, bind = null, unbind = null]
		Reference[name = EventAdmin, interface = org.osgi.service.event.EventAdmin, policy = static, cardinality = 1..1, target = null, bind = null, unbind = null]
	}
	located in bundle = org.goko.controller.tinyg_0.0.1.201509202039 [109]
] 
19:23:28.536 [org.goko.log.GokoLogConsoleReader@51a0aa70] INFO  OSGI Log -  [ Bundle = org.eclipse.equinox.ds] ErrorException occurred while creating new instance of component Component[
	name = org.goko.controller.tinyg.TinyGFeatureSet
	activate = activate
	deactivate = deactivate
	modified = 
	configuration-policy = optional
	factory = null
	autoenable = true
	immediate = false
	implementation = org.goko.controller.tinyg.TinyGFeatureSet
	state = Disposed
	properties = 
	serviceFactory = false
	serviceInterface = [org.goko.core.feature.IFeatureSet]
	references = {
		Reference[name = ISerialConnectionService, interface = org.goko.core.connection.serial.ISerialConnectionService, policy = static, cardinality = 1..1, target = null, bind = null, unbind = null]
		Reference[name = IGCodeService, interface = org.goko.core.gcode.service.IGCodeService, policy = static, cardinality = 1..1, target = null, bind = null, unbind = null]
		Reference[name = IGCodeExecutionMonitorService, interface = org.goko.core.gcode.service.IGCodeExecutionMonitorService, policy = static, cardinality = 1..1, target = null, bind = null, unbind = null]
		Reference[name = IApplicativeLogService, interface = org.goko.core.common.applicative.logging.IApplicativeLogService, policy = static, cardinality = 1..1, target = null, bind = null, unbind = null]
		Reference[name = EventAdmin, interface = org.osgi.service.event.EventAdmin, policy = static, cardinality = 1..1, target = null, bind = null, unbind = null]
	}
	located in bundle = org.goko.controller.tinyg_0.0.1.201509202039 [109]
] 
19:23:28.537 [org.goko.log.GokoLogConsoleReader@51a0aa70] INFO  OSGI Log -  [ Bundle = org.goko.core] Warn[SCR] Could not get the service object relevant to the reference. One possible reason is a circularity problem. Another possible reason is that BundleContext.getService() returns null.
	Details:
	Problematic reference = Reference[name = IFeatureSet, interface = org.goko.core.feature.IFeatureSet, policy = dynamic, cardinality = 0..n, target = null, bind = addFeatureSet, unbind = null]
	of service component = org.goko.core.feature.FeatureSetManagerImpl
	component implementation class = org.goko.core.feature.FeatureSetManagerImpl
	located in bundle with symbolic name = org.goko.core
	bundle location = reference:file:plugins/org.goko.core_0.0.2.201509202039.jar
19:23:28.544 [org.goko.log.GokoLogConsoleReader@51a0aa70] INFO  OSGI Log -  [ Bundle = org.eclipse.osgi] ErrorFrameworkEvent ERROR
1

cncgoko, 2015-12-29:
Hi,

I've run the install/update process and everything works fine.
What target board did you chose at first run ? 
Would it be possible to send your complete log files (using Gist maybe )? Maybe there is an error before this one that can explain. (both ```/log/*.log```  and ```/workspace/.metadata/.log``` files)

I think you already tried to restart it several times ? Maybe even try a download/update process again ?
Let's see with a more complete log first.

perfectphase, 2015-12-30:
Hi, I've tried re-downloading, running with the update and without, the 64 and the 32 bit versions (I'm on Win10 x64), latest version of Java installed.

I've attached the full log this time.

Thanks,

Stephen.

[log.2015-12-30.log.txt](https://github.com/cncgoko/Goko/files/74935/log.2015-12-30.log.txt)

cncgoko, 2015-12-30:
Even the release version from Github, out of the box (no update, selecting TinyG) doesn't work ?

perfectphase, 2015-12-30:
I'm using the zips that are downloaded from the readme.md at https://github.com/cncgoko/Goko, no changes. Unblocked then unzipped.

I can't select TinyG, the controller for it seems to fail to load (from looking at the log).  Only option I get is default or Grbl.

Java(TM) SE Runtime Environment (build 1.8.0_66-b18)
Java HotSpot(TM) 64-Bit Server VM (build 25.66-b18, mixed mode)

cncgoko, 2015-12-30:
I pushed a new update. Can you please update, restart Goko and send me back the log file ?
Also add the log file in the ```workspace/.metadata``` folder ?

perfectphase, 2015-12-30:
Hi, the app updated when I started.  Still no TinyG option, I've attached the log.

Looking at the new log, I'm on an en-GB version of windows.

[log.2015-12-30.log.txt](https://github.com/cncgoko/Goko/files/75240/log.2015-12-30.log.txt)
[.log.txt](https://github.com/cncgoko/Goko/files/75241/default.log.txt)

cncgoko, 2015-12-30:
Ok from the new log it looks like an I18N issue.
As a workaround, adding the following :
```
-nl
fr_FR
```
in the Goko.ini file should work. (It has to be on 2 separate lines, right before --launcher.library)

perfectphase, 2015-12-30:
Looks good!  I'm up and running!  Thanks.

cncgoko, 2015-12-30:
Feel free to open new issues if you find some. I'll fix the translation issue asap.

closed 2015-12-31 by cncgoko

cncgoko, 2015-12-31:
This issues is solved on last update.
You can remove the workaround in the .ini file.

commit 50cd5ebf (2015-12-31): Fix #21 TinyG support - I18N file issue