cncgoko/Goko #29: Option to invert orbit controls in gcode viewer

turpinator, 2016-09-18:
Not sure if youre continuing dev, see the last commit was ~4 months ago, but might as well suggest nonetheless.

Add an option to invert the orbit controls in the gcode viewer, as it stands, it is completely inverse of every single program with ant sort of 3D view that Ive ever seen.

Anyway, this looks awesome, Id love to see it completed/I dont know.  I just know I dont care for Chilipeppr.

cncgoko, 2016-09-18:
@turpinator Hello

The dev is still going on :) mainly on the dev branch and the next release is planned anytime soon. And it's pretty cool to see some requests here. I gives me motivation for the devs.

Your request is a good idea, I'm used to inverted mouse (in CAD, games...).
Just to be clear, when you wirte "an option to invert the orbit controls", what does that mean ?
- Invert the used mouse button ? (left click = orbit, and right click = move)
or
- Invert the response of the orbit to the mouse moves ? 

Can you please give more detail about the expected behavior of the camera given the mouse motion ?

Thanks,
Psyko

turpinator, 2016-09-18:
Invert the response.  So it will be like this.  http://i.imgur.com/lKLLhEw.gifv  Mouse goes right, object rotates counterclockwise (from above).   Hard to explain, I guess.

As for the mouse buttons, i guess it is a little reversed than what im used to, but thats easier to relearn than the orbit directions.

cncgoko, 2016-09-18:
I see. This should not be a problem.

closed 2016-10-04 by cncgoko

cncgoko, 2016-10-04:
Preferences panel is available in latest release (0.3.1)

turpinator, 2016-10-07:
Had a chance to try it today, exactly what I was thinking. :+1:

cncgoko, 2016-10-08:
@turpinator 
Thanks for the feedback :)

commit 84cd11ea (2016-09-19): Fix #29