cncgoko/Goko #32: Stuttering while moving

hk89, 2017-01-08:
Hi,

I have noticed lately that the machine has a small stuttering while moving through arcs. When I jog the machine, I notice that the movement is not fluid as before the last update. This also happens while executing any type of gcode; there are small stutters, nearly less than a second, through an arc.

This happened when GOKO asked me to overwrite some settings in GRBL for it work properly right after the update. Do not know what is going on here.

cncgoko, 2017-01-08:
Hi Hk89,

When you say arc, arc they GCode arc using G2/G3 or arc defined by multiple small linear G1 motion ?
In this latest case, it might happens. 

Goko asked to change the configuration when State report mask does not include Planner buffer report.

Could you please provides one GCode file that causes the issue ?

hk89, 2017-01-14:
I checked the gcode file and I have found that it uses the G2 and G3 commands for arcs. Here is the file.
[vendetta.zip](https://github.com/cncgoko/Goko/files/706029/vendetta.zip)

cncgoko, 2017-01-14:
While investigatin, I noticed that my $11 setting on Grbl went to 0.000
It should be something around 0.020 (as mentioned [here](https://github.com/grbl/grbl/wiki/Configuring-Grbl-v0.9#11---junction-deviation-mm))

Until I identify the issue and publish a fix, you wan't to make sure that your param $11 is not zero by issuing a $11=0.020 command in the serial console while connected. That should do the trick temporarly.

commit 278812fb (2017-01-15): Fix #32
closed 2017-01-15 by cncgoko