cncgoko/Goko #33: Saving and executing Macros

hk89, 2017-01-14:
Hi,

When I tried to use the Macro feature for the first time, I could not edit its name because all buttons were disabled while typing the new name and there was no way for me to save it. So, I was stuck with a Macro that is called "New Macro". Another issue is that the Macro was edited and saved, but could not able to execute it.

I done a Macro with G0 X0 Y0 Z0 and simply could not execute it.

Maybe, I am messing something here.

cncgoko, 2017-01-14:
Hi,

You have to create a macro by pressing the "New macro" button, and then click "Edit".
Maybe the "Edition mode" should be enabled when the macro is created.
Anytime you want to change the macro (including the associated GCode) you have to press the edit button, above the list of macro.

The documentation is available here http://docs.goko.fr/
The section about "working with macros" explains it : http://docs.goko.fr/master/documentation/index.html#_working_with_macros

Let me know if the documentation requires more explanation.

cncgoko, 2017-03-02:
Documentation was updated : http://docs.goko.fr/master/views/macro.html

Let me know if it's better

hk89, 2017-03-10:
Documentation is clear, however I can not add another macro or edit/delete old macros after adding a first one because the 'New' and 'Edit' buttons are disabled. Using ver. 0.3.3

Also, macros are not regained after GOKO restart

cncgoko, 2017-03-11:
IF the 'New' and 'Edit' button are disabled, it means you didn't click the 'Apply' or 'Cancel' button on the current macro to validate your changes. Once a macro is created, it automattically enters the "Edit mode" which requires you to applyu or cancel the changes you've made.

You says 'macros are not regained after GOKO restart'.  That means at each start your list of macro in the macro manager is empty ?

That's weird, because I cannot reproduce the error. Can you please post the 2 logs files from :
- **GOKO_FOLDER/log/** _(find the most recent file)_
- **workspace/.metadata/.log** file

And send them to me via email ?

Also, check the content of your **workspace/macros/** folder. Does the folder exist ? Is there any file in it ?

Thanks,