cncgoko/Goko #34: Console filtering feature question

hk89, 2017-01-14:
Hi,

I love the new feature of filtering the console. However, I do not know how to use arguments. Is there any thing I can read somewhere to help me? I could not find it anywhere. Maybe using predefned buttons for these arguments that a user can click on to add them in the "Parameter" box would be quite handy.

cncgoko, 2017-01-14:
Hi,

The documentation is available here http://docs.goko.fr/
The section about serial console explains it : http://docs.goko.fr/master/documentation/index.html#_serial_connection

Let me know if the documentation requires more explanation.

hk89, 2017-03-10:
Nice documentation! Thanks!

I guess I have to learn those patterns for some time! :D

closed 2017-03-11 by cncgoko