cncgoko/Goko #36: Issues upon start

hk89, 2017-02-06:
Hi, 

A couple of times I run Goko and connect, I find that my G54 origin is thrown away to the bottom left of the top view away from the normal origin; in the - X and - Y region. Temporary workaround I found is to reset the EEPROM by sending a command $RST=#.

Also, upon connecting, sometimes I find that the machine position is very far away through X and Y; not even close to the origin. I have to zero it out everytime that happens.

cncgoko, 2017-02-07:
Hi,

Adter investigation, I identified an issue in the management of coordinates system, but I'm not sure it's related to your issue.
Do you have any protocol to reproduce the issue ? 
How do you set the coordinate system offset ? Using the command panel, or by issuing a GCode command in the console ?

hk89, 2017-03-10:
That fixed my issue. Thanks for the kind support.

closed 2017-03-11 by cncgoko