cncgoko/Goko #40: Error when connecting

hk89, 2017-06-02:
Hi,

I found that when connecting to grbl 1.1 on a nano, Goko (0.3.4) was not able to communicate with it. DRO's state is 'undefined', queue can not be started, I can not send serial commands and command panel buttons are grayed out. See attached image.

I did not change any of the serial communication preferences.
[image]

cncgoko, 2017-06-02:
Hi, 

What's the value of Flow Control ? 
It should be turned off for Grbl.

hk89, 2017-06-02:
I believe that I freshly flashed grbl 1.1 on a new nano with blank EEPROM, so I think that the flow control is off by default. Am I right?

cncgoko, 2017-06-02:
Yes but did you turn it off in Goko as well ?

hk89, 2017-06-02:
I don't know what happened but I found that it was set to "RTS/CTS". I don't remember changing that!

I set it to XON/XOFF and now running!

cncgoko, 2017-06-02:
You'd better turn it off completely ! I think Grbl does not support flow control.

closed 2017-06-02 by cncgoko