cncgoko/Goko #44: Grbl 1.1f not recognised

silicontrip, 2017-08-03:
I have Grbl 1.1f running on my arduino.
Goko will connect to it and I see the Grbl prompt

``Grbl 1.1f ['$' for help]``

 in the terminal window, but Goko doesn't allow any further operations with it.

As an aside I can't sign up to the discussion page.  I was going to report this issue there.
And I'm also having trouble building it within Maven.  To attempt to resolve this issue and the GL3 issue.  I wouldn't mind discussing these issues in the community.

Mark

cncgoko, 2017-08-04:
Hello Mark,

I'm currently on vacation, so I won't be able to take a look for the next 2
weeks.
What kind of issue do you have when connecting to discuss.goko.fr ?
Are you creating a brand new account, or using a google, github or twitter
account ?

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Garanti
sans virus. www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

cncgoko, 2017-08-17:
Hi @silicontrip,

I finally have access to my Grbl board again. It runs Grbl 1.1f and connects perfectly with Goko.
Can you please check your Serial connection settings ?
I use the following settings in Goko:
- Baudrate: 115200
- Data bits: 8
- Parity: None
- Stop bits: 1
- **Flow control: None**

The misconfigured flow control can give unexpected results like the one you're having.
Let me know if it worked,

silicontrip, 2017-08-18:
Ah, interesting, flow control defaults to RTS/CTS
Turning this off resolved the comms issues.

Thanks

closed 2017-08-18 by cncgoko