cncgoko/Goko #52: Archlinux 64bit SISEGV

emogg, 2018-03-05:
I am getting this error on an archlinux 64bit system. Any ints as to the problem?

(Goko:10227): Gtk-CRITICAL **: gtk_widget_get_style_context: assertion 'GTK_IS_WIDGET (widget)' failed

(Goko:10227): Gtk-CRITICAL **: gtk_style_context_add_provider: assertion 'GTK_IS_STYLE_CONTEXT (context)' failed

# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGSEGV (0xb) at pc=0x00007f52de1562d7, pid=10227, tid=0x00007f531f87e700
#
# JRE version: OpenJDK Runtime Environment (8.0_162-b01) (build 1.8.0_162-b01)
# Java VM: OpenJDK 64-Bit Server VM (25.162-b01 mixed mode linux-amd64 compressed oops)
# Problematic frame:
# C  [libgtk-3.so.0+0x1882d7]  gtk_css_provider_load_from_data+0x27

emogg, 2018-03-05:
Sorry about the bold text!

cncgoko, 2018-05-15:
Hello @emogg 

Can you please try and install Eclipse Mars and run it ?
Goko uses Eclipse Mars framework, so I would like to know if it's related to the framework, or specifically to Goko.

Let me know how it works.