cncgoko/Goko #54: Problems starting on MacOSX

breiler, 2018-06-29:
I have problems starting Goko on a Mac where the first window won't allow me to select controller. The drop down menu and the ok/cancel buttons are unresponsive and nothing happens when I click on them.

Also tried cloning the project, building and running it with the latest source with the same result.

[image]

cncgoko, 2018-08-21:
Hello @breiler,

Do you have anything in the logs ? Can you send them ?
Looks for .log files in the Goko directory (log/ and workspace/.metadata/.log)

Thanks,

cncgoko, 2018-08-22:
Can you paste the .metadata/.log please ?

breiler, 2018-08-22:
Oh sorry, I missed that:

```
!SESSION 2018-08-22 06:23:56.448 -----------------------------------------------
eclipse.buildId=0.3.5-2017-11-04 10:36:28
java.version=9.0.4
java.vendor=Oracle Corporation
BootLoader constants: OS=macosx, ARCH=x86_64, WS=cocoa, NL=en_SE
Framework arguments:  -equinox.ds.debug=true -equinox.ds.print=true -keyring /Users/zefal/.eclipse_keyring -showlocation
Command-line arguments:  -os macosx -ws cocoa -arch x86_64 -equinox.ds.debug=true -equinox.ds.print=true -keyring /Users/zefal/.eclipse_keyring -showlocation

!ENTRY org.eclipse.equinox.ds 1 0 2018-08-22 06:23:58.096
!MESSAGE Could not bind a reference of component org.goko.core.feature.FeatureSetManagerImpl. The reference is: Reference[name = IFeatureSet, interface = org.goko.core.feature.IFeatureSet, policy = dynamic, cardinality = 0..n, target = null, bind = addFeatureSet, unbind = null]

!ENTRY org.eclipse.equinox.ds 1 0 2018-08-22 06:23:58.153
!MESSAGE Could not bind a reference of component org.goko.core.xmlPersistenceService. The reference is: Reference[name = IXmlPersistenceProvider, interface = org.goko.core.common.io.xml.IXmlPersistenceProvider, policy = dynamic, cardinality = 0..n, target = null, bind = addXmlPersistenceProvider, unbind = null]

!ENTRY org.eclipse.equinox.ds 1 0 2018-08-22 06:23:58.176
!MESSAGE Could not bind a reference of component org.goko.core.workspace.service.MapperServiceImpl. The reference is: Reference[name = ILoader, interface = org.goko.core.workspace.service.ILoader, policy = dynamic, cardinality = 0..n, target = null, bind = addLoader, unbind = null]

!ENTRY org.eclipse.equinox.ds 1 0 2018-08-22 06:23:58.176
!MESSAGE Could not bind a reference of component org.goko.core.workspace.service.MapperServiceImpl. The reference is: Reference[name = IExporter, interface = org.goko.core.workspace.service.IExporter, policy = dynamic, cardinality = 0..n, target = null, bind = addExporter, unbind = null]

!ENTRY org.eclipse.equinox.ds 1 0 2018-08-22 06:23:58.177
!MESSAGE Could not bind a reference of component org.goko.core.workspace.service.MapperServiceImpl. The reference is: Reference[name = IMapperProvider, interface = org.goko.core.workspace.service.IMapperProvider, policy = dynamic, cardinality = 0..n, target = null, bind = addMapperProvider, unbind = null]

!ENTRY org.eclipse.equinox.ds 1 0 2018-08-22 06:23:58.178
!MESSAGE Could not bind a reference of component org.goko.core.workspace.service.WorkspaceService. The reference is: Reference[name = IProjectSaveParticipant, interface = org.goko.core.workspace.service.IProjectSaveParticipant, policy = dynamic, cardinality = 0..n, target = null, bind = addProjectSaveParticipant, unbind = null]

!ENTRY org.eclipse.equinox.ds 1 0 2018-08-22 06:23:58.179
!MESSAGE Could not bind a reference of component org.goko.core.workspace.service.WorkspaceService. The reference is: Reference[name = IProjectLoadParticipant, interface = org.goko.core.workspace.service.IProjectLoadParticipant, policy = dynamic, cardinality = 0..n, target = null, bind = addProjectLoadParticipant, unbind = null]

!ENTRY org.eclipse.equinox.ds 1 0 2018-08-22 06:23:58.188
!MESSAGE Could not bind a reference of component org.goko.core.execution.monitor. The reference is: Reference[name = IGCodeProviderRepository, interface = org.goko.core.gcode.service.IGCodeProviderRepository, policy = dynamic, cardinality = 0..n, target = null, bind = addGCodeRepository, unbind = null]

!ENTRY org.eclipse.equinox.ds 1 0 2018-08-22 06:23:58.232
!MESSAGE Could not bind a reference of component org.goko.core.execution.monitor.ExecutionQueueContainerUiProvider. The reference is: Reference[name = IExecutionQueueContributionItem, interface = org.goko.core.execution.monitor.uiprovider.menu.executionqueue.IExecutionQueueContributionItem, policy = dynamic, cardinality = 0..n, target = null, bind = addExecutionQueueContributionItem, unbind = null]

!ENTRY org.eclipse.equinox.ds 1 0 2018-08-22 06:23:58.301
!MESSAGE Could not bind a reference of component org.goko.core.workspace.service.WorkspaceUIService. The reference is: Reference[name = ProjectContainerUiProvider, interface = org.goko.core.workspace.bean.ProjectContainerUiProvider, policy = dynamic, cardinality = 0..n, target = null, bind = addProjectContainerUiProvider, unbind = null]

!ENTRY org.eclipse.equinox.ds 1 0 2018-08-22 06:23:58.328
!MESSAGE Could not bind a reference of component org.goko.core.gcode.rs274ngcv3. The reference is: Reference[name = IGCodeProviderRepositoryListener, interface = org.goko.core.gcode.service.IGCodeProviderRepositoryListener, policy = dynamic, cardinality = 0..n, target = null, bind = addListener, unbind = null]

!ENTRY org.eclipse.equinox.ds 1 0 2018-08-22 06:23:58.424
!MESSAGE Could not bind a reference of component org.goko.gcode.rs274ngcv3.ui. The reference is: Reference[name = IModifierUiProvider, interface = org.goko.gcode.rs274ngcv3.ui.workspace.uiprovider.IModifierUiProvider, policy = dynamic, cardinality = 0..n, target = null, bind = addModifierBuilder, unbind = null]

!ENTRY org.eclipse.equinox.ds 1 0 2018-08-22 06:23:58.449
!MESSAGE Could not bind a reference of component org.goko.gcode.rs274ngcv3.ui.GCodeContainerUiProvider. The reference is: Reference[name = IGCodeProviderContributionItem, interface = org.goko.gcode.rs274ngcv3.ui.workspace.uiprovider.menu.gcodeprovider.IGCodeProviderContributionItem, policy = dynamic, cardinality = 0..n, target = null, bind = addGCodeProviderContributionItem, unbind  [... cut ...]
	at goko.GokoLifeCycleManager.postContextCreate(GokoLifeCycleManager.java:78)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.base/java.lang.reflect.Method.invoke(Method.java:564)
	at org.eclipse.e4.core.internal.di.MethodRequestor.execute(MethodRequestor.java:56)
	... 17 more
```

cncgoko, 2018-08-22:
I don't see any unexpected behaviour, except the NPE, but I guess you hit the OK button with nothing in the combobox, right ? 

Any way to do a live debug ? If so, send me an email at contact@goko.fr

breiler, 2018-08-22:
No it's not possible to do a live debug. 

The buttons is not clickable. It's like the splash is modal even though the "Select your machine"-dialog is shown on top:

[image]

breiler, 2018-08-22:
Good test, it seems that the exception occurs when closing the application.

cncgoko, 2018-08-22:
Ok, so my guess is that the combo is nicely populated, but a focus issue prevents you from selecting anything in it.

One workaround that might work is to go in the folder :
```workspace\.metadata\.plugins\org.eclipse.core.runtime\.settings```

and modify or create the ```org.goko.core.prefs``` file. There is a target board value that you can set.
Mine looks like: https://gist.github.com/cncgoko/bb31f42dbc71b6ed8d5d2dbc5fc203c3

Valid values for target board parameters are:
- For G2: ```g2core.v09```
- For Grbl:  ```grbl.v08```  or  ```grbl.v09```  or  ```grbl.v11```
- For TinyG:  ```tinyg.v097```

breiler, 2018-08-22:
That works for getting the software to start, unfortunately the same focus issue exists for the rest of the application once it's loaded.

cncgoko, 2018-08-22:
Yes, I was kind of expecting that... Will take a look asap

cncgoko, 2018-08-22:
Last try, if you have it running in Eclipse, try commenting lines 53 and 54 and 98 in files ```GokoLifeCycleManager.java```

```
final GokoSplashscreen splash = new GokoSplashscreen();
splash.open();
```

I have no other clues than that. I'll have to set up a test environment.

breiler, 2018-08-22:
I feel like a noob... =(
How do I import the project in Eclipse? Or even better, how do I build it using maven?

cncgoko, 2018-08-22:
Well, don't feel like a noob. The setup is really not easy, and it took me a LOT of time to do it.
It's too long to explain here. Maybe I will write a quick tutorial on that someday. 

For now, I'll setup a MacOS environment. Always usefull.

cncgoko, 2018-08-24:
@breiler I manage to solve the issue by using Java JDK 1.8.151. Multiple java application report the same issues. Unfortunatly, so far I couldn't find anyone who clearly identified the issue.
The 3D preview doesn't work, but I can't tell if it's because of the virtual machine or not.
For the context, can you tell me what MacOS and java version you are using when it doesn't work?

breiler, 2018-08-24:
I'm using MacOSX 10.12.6.

According to the metadata log Goko is running Java 9.0.4 which I also have installed but is not my default. I tried to remove that Java version (simply by just deleting it) and Goko started using my Java 1.8.0_60 which works. Thanks!

I have the same problem with the 3D preview though.

cncgoko, 2018-08-24:
No Goko is supposed to run on Java 8, not 9.
I will have to check that 3d preview issue... and still, the source problem is not solved. We only found a workaround