cncgoko/Goko #55: Not able to control anything after connecting

jds598, 2018-10-10:
I'm not sure if this is a problem with goko, the tinyg2 fw, or my own ignorance. I've got the latest tinyg2 flashed onto my due. I currently don't have anything hooked up to the due other my host computer, I don't know if that is my problem but I'm coming from grbl and I know that will connect and show info when nothing else is connected. So when I connect to the due in goko it shows up in the lower left corner that its connected on com10 at 115200 but no controls are available and in the g2 configuration menu it has 0 for firmware version and all other fields,also the serial monitor shows the info that is being pushed by goko but I don't believe its receiving anything back. I've tried installing g2 a few times on 2 different boards with no luck. Please help!!

justinclift, 2018-10-10:
Ahhh this sounds familiar.  As a thought, try flashing the [older 100.26 version](https://github.com/synthetos/g2/releases/tag/100.26) of g2 to the board and see the problems go away.

It's been a while since I looked at this personally, but I remember having success with 100.26 (I think), but hit some problem with the later (edge) g2 versions.  This is from possibly dodgy memory though (!), so not 100% certain.

jds598, 2018-10-10:
Thanks in advance. Where can I download the 100.26 version. And by dodgy memory I take it you mean the sticks of ram installed in my pc correct?

justinclift, 2018-10-10:
Ahhh no.  I'm meaning literally the memory in my head, as I haven't done much with g2 recently (been moving house instead). :smile:

The 100.26 version of g2 is here: https://github.com/synthetos/g2/releases/download/100.26/g2core-gShield-100.26.bin

In theory, with that you should be good to go.  If you still hit issues, try either CNCjs or Universal G-Code Sender (nightly builds), as they're both known to work with 100.26 "for sure":

* https://github.com/cncjs/cncjs
* https://github.com/winder/Universal-G-Code-Sender#downloads  (Pick the "UGS Platform" one there)

I'm about to get some sleep, so if you do get stuck don't be too stressed if you don't hear from me right away. :wink:

jds598, 2018-10-10:
I just moved myself not to long ago, I'm glad its done. No worries if it takes you some time to respond, I'm not pressed for time with this at all. I appreciate the help.

So far still no luck. I downloaded the 100.26 version and flashed it to both of my dues and still the same results. I also tried cncjs and ugs platform nightly build. If I go to machine-firmware settings in ugs it gives an error msg saying no firmware found. I tried my grbl board with ugs and it connected and controlled everything just fine.

 I did notice in device manager that neither due ever shows up as tinygv2 and there's only one port linked with the board, not 2. I'm not supposed to have both native and programming port hooked up at the same time, am I? I've tried both ports but not at the same time, I'm assuming its the native port that's used for connecting.

  At one point I hit the reset button on the due and windows said something to the effect of configuring device tinygv2, but when it was done its now named usb serial device in device manager. 

I'm stumped :(

jds598, 2018-10-11:
One other question, does there have to be anything other than a usb cable hooked to the due. I don't have any sort of shield on it or any other power aside from the usb.

jds598, 2018-10-11:
I watched the video, the thing that is confusing me is that in the video it shows in device manager there are 2 coms associated with tinyg but in my device manager its only coming up with one and its not called tinygv2. Do you think I would have better luck setting up a linux machine and trying from that? I have enough spare parts to put another computer together.

justinclift, 2018-10-12:
Hmmm, going the Linux approach won't hurt, but we really should be able to get it working on Windows.

When you plug the Due in on windows, what do the ports show up as in Device Manager?

As a data point, I remember when starting out I tried Windows too, and wasn't sure if I needed to plug in two cables at once, or just one cable, and if one cable then which port?

With some experience, I've found I only need the one port (Native port serial USB) connected to do flashing and/or comms with, so that's the only one I hook up now.

[image]

Does that help? :smile:

jds598, 2018-10-12:
When I plug into the Due's programming port it shows up as arduino due programming port. When I plug into the Due's native port, when there's no firmware loaded it shows up as bossa programming port, when tinyg2 is on the Due it shows up as serial communication port. I can't program it with bossac.exe over the native port, it give me Sam ba error

jds598, 2018-10-12:
So this is a shot of device manager with the driver details. Its only coming up with one port for the Due. I tried using the g2 data driver, and then the control channel driver but it didn't make a difference, although its probably because I cant use both at the same time on one port. Every time I flash the Due with new g2 firmware windows automatically assigns that usbser.sys driver. I went and removed the usbser.sys driver from my system32 folder thinking that would stop it from happening and then maybe it would show up with 2 ports but it keeps reassigning that driver, I don't know where its pulling it from. I even disconnected from the internet thinking it was downloading it but windows still assigned it somehow.  
[image]

This is from goko serial monitor with the printrbot firmware. It seems to be connected but I don't know how to read the code and know what all that means.

[image]

This is from goko serial monitor after I use some of the jog buttons.

[image]

When I do use any of the controls in goko none of the communication led's on the Due flash at all, are they supposed to? Would you know where I could get the pin assignment for printrbot board

At this point I'm thinking it is an issue with the drivers, I just am at a loss with how to get the Due to show up with 2 ports in device manager.

justinclift, 2018-10-12:
Ahhh.  I'm not sure what's going on with the drivers, but the communication stuff you're getting back from the Due is looking ok.

The stuff you're getting back is in JSON format.  It's optimised for machines to read easily, and isn't real human friendly (as you're noticing too). :wink:

&nbsp; &nbsp; https://github.com/synthetos/g2/wiki/Text-Mode#text-mode-and-json-mode

I don't remember if goko lets you type stuff into that serial output console area (it's been ages for me and I don't have my electronics stuff with me atm. :frowning: )

If it **does** let you type stuff in, then try switching the output to text mode.  You should be able to do that by typing in:

    $ej=0

After which the status reponses back should make more sense. :smile:

That setting doesn't stick in the Due though, so every time it gets restarted it'll come up in JSON mode.

Thinking about this more, I do reckon you'll be better off to set up a Linux machine.  That's because the Due seems to be doing ok, and it does look like the driver problem is the weirdness.  On Linux, in theory things should "just work" fine.

That being said, how experienced with Linux are you? :smile:

jds598, 2018-10-13:
I don't have a whole lot of experience with Linux. I played around with ubuntu awhile back. I don't believe it would be to hard to figure it out enough for what I need. I cant write code of any sort so that aspect wont make a difference for me.

 I did just get a key for windows 7, I'm gonna give that a try first just because I know windows pretty well and it seems that tinyg2 was working on windows7 for sure. If I still have problems then I will switch to Linux. what version of Linux would you recommend?

Thanks again for the help!!

justinclift, 2018-10-13:
No worries at all. :smile:

With Linux, I'd recommend the KDE (specifically) version of CentOS 7.

&nbsp; &nbsp; http://isoredirect.centos.org/centos/7/isos/x86_64/CentOS-7-x86_64-LiveKDE-1804.iso

CentOS 7 is good because it's super stable + gets security updates, so you can use the same installation for (literally) years without it becoming insecure.  The KDE version of that because KDE's pretty good in general (my opinion), and it's suitable for ex-windows people too.  eg start button in the bottom left, taskbar, etc. :smile:

justinclift, 2018-12-05:
No worries at all.  I definitely know what that's like. :grin:

jds598, 2018-12-05:
Sorry I haven't updated this in awhile. I haven't solved the issue, I've been busy with work and haven't had the time to continue troubleshooting this. I'm at a loss as to what to try next with it. Thanks.

cncgoko, 2018-12-05:
From what I see in https://github.com/cncgoko/Goko/issues/55#issuecomment-429464285 there is no issue with your board and Goko. Goko is able to communicate nicely with it and you should be able to control your machine using Goko. There might be issues with G2 incompatible version, but it should be mostly ok unless they changed communication protocol.

Let me know if that's correct so I can close the issue.

justinclift, 2018-12-05:
> ... unless they changed communication protocol.

Don't rule that out. :wink:

There's definitely been talk about doing so, but I don't personally remember if it's been put into action yet.  Also, the "edge" (development) releases of g2core have some bugs in them compared to the "master" (stable) branch.  Wouldn't recommend using the edge version until those stabilise a bit and the corresponding GUI's (Goko, UGS, CNCjs, etc) have a chance to adapt for any changes (!).

jds598, 2018-12-05:
I have tried a few different releases and all have the same outcome. I don't think it's a problem with goko because the behavior is the same with ugs and cncjs. Maybe I will put a post up on the tinyg2 page. If I get a chance this weekend I will give it another try. Also I could be using the wrong pins as I am making an educated guess on wich pins are what by using the example pinout. I may also give a go at compiling it myself so I know what pins to use.