cncgoko/Goko #57: feature request: 5th axis

Dirk--, 2018-12-08:
Thanks for your work!

Any plans to add a fifth axis?

Thanks, Dirk

Dirk--, 2018-12-08:
having 0.3.5 on my win64-machine, I see only 'A' on Command panel and can add 'A' only to DRO ...

cncgoko, 2018-12-08:
You're correct. It's not available in those panels. 
What controller do you use ? G2 ? TinyG ? Grbl ? Something else ?

Dirk--, 2018-12-08:
right now, just doing some research. g2core will be set up next week ...
https://www.djuke.nl/index.php?page=shop.product_details&category_id=1&flypage=flypage.tpl&product_id=406&vmcchk=1&option=com_virtuemart&Itemid=8

cncgoko, 2018-12-08:
Adding B & C axis in DRO and Command panel is no big deal so yes it can be done easily. 
Displaying it in the 3d preview might be difficult :) Not sure how to display it, since 5th axis could be used for different purpose.

Dirk--, 2018-12-08:
this might be of interest:
- https://github.com/winder/Universal-G-Code-Sender/issues/1068
- https://github.com/cncjs/cncjs ( do not know anything about viewer's capabilities, at least 6-axis support at DRO and command-panel)
- http://linuxcnc.org/docs/devel/html/motion/5-axis-kinematics.html